flo199213/Hoverboard-Firmware-Hack-Gen2 #2: #include "arm_math.h"

drbytes, 2018-09-01:
Hi, I installed all the relevant software packs in Keil uVision ... at least that what I thought but I get a compilation error in main.c around line 
`47: #include "arm_math.h" `
resulting in 
```
Src\main.c(47): error:  #5: cannot open source input file "arm_math.h": No such file or directory
  #include "arm_math.h" 
```

Do you know what I'm missing? (keil n00b)

drbytes, 2018-09-01:
I've commented the line and it builds.

Johnvan, 2019-02-24:
I have a similar problem with an adafruit MP3 program that uses arm_math.h, what do I have to do to fix this thing.
ERROR: arm_math.h: No such file or directory

pieterjanbuntinx, 2019-02-26:
It now works for me. To solve the other error I got I had to enable this:
[image]

Johnvan, 2019-02-28:
I have no idea what you are trying to tell me about **Enabling this,** what and how. Where is this Options for Target.

gaucho1978, 2019-04-19:
> You have to include the path C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.4.0\CMSIS\Include in the "Options for Target..."  > C/C++. Afterwards it compiles al the c-files but it gets stuck on the other issue that's reported in this repo.

in my case the correct folder is: C:\Users\GauchoHP\AppData\Local\Arm\Packs\ARM\CMSIS\5.5.1\CMSIS\Include

after doing this it correctly compile.

renmalek7, 2020-04-08:
Hi. Were you able to flash this firmware?

Johnvan, 2020-04-09:
I updated my IDE and the fault went away. Thankyou for your assistance however, much appreciated.

weiminshen99, 2023-05-19:
The <arm_math.h> and other related .h files may also be found at this repository:
https://github.com/PaulStoffregen/arm_math