flo199213/Hoverboard-Firmware-Hack-Gen2 #3: Hoverboard.axf not found

bugiworks, 2018-11-03:
I appreciate your work, after 1 board hoverboard, i want to try this one also. 

when i try to build with keil, i get the error
.\Objects\Hoverboard.axf: error: L6031U: Could not open scatter description file .\Objects\Hoverboard.sct: No such file or directory
Not enough information to list image symbols.
Not enough information to list load addresses in the image map.
Finished: 2 information, 0 warning, 0 error and 1 fatal error messages.
".\Objects\Hoverboard.axf" - 1 Error(s), 0 Warning(s).
Target not created.

any solution? or any direct hex files do you have?
By the way, the board im using has GD32F103 not F130, i have downloaded the pack for 103 but im not sure if it works same.

pieterjanbuntinx, 2018-12-21:
Hi @kevin0932,
No, I did not. But I found [a fork](https://github.com/btsimonh/Hoverboard-Firmware-Hack-Gen2) of this repo that compiles. I have not managed to get it uploaded to my hoverboard controller, I don't have a STLink but I'm using a STM32 Discovery board, so that's probably the problem.

bugiworks, 2018-12-25:
Unfortunately i couldnt find any solution.

pieterjanbuntinx, 2019-02-26:
It now works for me. To solve this error I had to enable this: 
[image]

renmalek7, 2020-04-08:
Hi. Were you able to flash this firmware?

pieterjanbuntinx, 2020-04-08:
Yes, I was able to flash it. It works great. What are you stuck on?

renmalek7, 2020-04-08:
I have a different board, it's also a GD32F130. Before flashing the firmware, How can I tell if the Pins are the same?
[image]

pieterjanbuntinx, 2020-04-08:
You could perhaps measure continuity using a multimeter from the pins of the MCU to the peripherals. If that corresponds to the pinout in the code it should work. That will be quite some work though, so if you think that the layout of the PCB looks about the same as the one of this repo, you could perhaps just try it out (at your own risk). Replacement boards are not that expensive from AliExpress

renmalek7, 2020-04-08:
ok, I will do that and post the results. Do you know if there is a way to pull the original firmware from the board? Would it be posible to write it back in the board if I fail and its not burned l?

pieterjanbuntinx, 2020-04-08:
No I don't think so. I thought it was locked or something. It has something to do with the options bytes

renmalek7, 2020-04-08:
ok, thank you very much for you time : )

Harsh2210, 2020-06-16:
> I have a different board, it's also a GD32F130. Before flashing the firmware, How can I tell if the Pins are the same?
> [image]

Were you able to flash the firmware? I am using the same motherboard. I wanted to know whether the pinouts are same for this one.

yodaco, 2022-07-23:
> > I have a different board, it's also a GD32F130. Before flashing the firmware, How can I tell if the Pins are the same?
> > [image]
> 
> Were you able to flash the firmware? I am using the same motherboard. I wanted to know whether the pinouts are same for this one.

Did it work?!?