flo199213/Hoverboard-Firmware-Hack-Gen2 #5: Restore hoverboard

gaucho1978, 2019-04-13:
Is it there a way to restore the hoverboard after playing with this hack?

teixeluis, 2019-06-21:
That is a pertinent question. Anyone have been successful at dumping the original firmware via ST-Link? Is it locked in most of not every board?

If a dump of the firmware + eeprom + config bits is feasible, eventually the device could be restored (I would put very little hope that the device flash is not protected though..). 

Cheers,

Luis Teixeira

renmalek7, 2020-04-06:
Anyone did it?

renmalek7, 2020-04-08:
Hi. Were you able to flash this firmware?

weiminshen99, 2023-05-19:
I have saved the original flash for both Master and Slave board, if anyone is still interested.

teixeluis, 2023-05-19:
@weiminshen99  cool. Yes I would have interest in taking a peek. Cheers

weiminshen99, 2023-05-19:
The saved .bin files are at https://github.com/weiminshen99/GD32-hover-master-slave.git, under the directory Factory_Flash.

I have saved them after I unlocked boards, but I never test them because my boards are all occupied for other purposes. Good luck, and let me know if they work or not. Thanks!