flo199213/Hoverboard-Firmware-Hack-Gen2 #10: Please add a makefile for my online compiler :-)

RoboDurden, 2019-12-30:
Hello from Germany :-)
I would like to add your repo to my online compiler www.pionierland.de/hoverhack
videos: https://www.youtube.com/playlist?list=PLa9VLvJBLbTMmpslED2FMClIBv5URBMIa

But you only offer the Keil project files. If you are already compiling with GCC, a port should be possible: https://community.arm.com/developer/tools-software/oss-platforms/f/gnu-toolchain-forum/45142/how-to-move-from-keil-to-linux

Ideas welcome !
www.robo4future.de :-)

RoboDurden, 2020-01-14:
no answer :'(

renmalek7, 2020-04-06:
Hello. Were you able to move the repo to your online compiler? I am working with this firmware and I would appreciate your advice. Could you please mail me to renmalek7@gmail.com

Thank you very much : )

RoboDurden, 2020-04-06:
No, i am not experienced enough to copy and paste a Makefile :-/
A pity because adding repos to my online compiler just takes one line of code. Today i added all three repos of  https://github.com/EmanuelFeru

renmalek7, 2020-04-06:
What about Florian's firmware (this branch)? Were you able to flash it in a Gen2 board?

RoboDurden, 2020-04-06:
I have no Gen2 board where there is a control board for either side.

renmalek7, 2020-04-06:
Do you have 20 min for a video call? I am having trouble with the firmware and maybe you can advise me. Do you have skype?

renmalek7, 2020-04-08:
please, I just need someone to tell me how to indentify the pin conection in the board : (

RoboDurden, 2020-04-08:
You are spoliing my issue here to kindly ask for a makefile.

renmalek7, 2020-04-08:
I am really sorry. Thats why I left my email. Could you plase help me?

bricobi5, 2022-06-05:
Hi I have 4 gen2 motherboard that I can't flash so I try Florian's firmware 
but where can i find it?

RoboDurden, 2022-06-05:
https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2/tree/master/HoverBoardGigaDevice
But as it still has no makefile, i can not include it to my online compiler :-(
I guess you have to install Keil to open the .uvoptx µVision 5 project files :-((
Someone who does not dislike Linux so strongly as i do might easily generate that makefile for me/us :-/

BastelPichi, 2022-07-23:
> https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2/tree/master/HoverBoardGigaDevice But as it still have no makefile, i can not include it to my online compiler :-( I guess you have to install Keil to open the .uvoptx µVision 5 project files :-(( Someone who does not dislike Linux so strongly as i do might easily generate that makefile for me/us :-/

Incoperating #16 may fix the issue. Compiling with PIO should be possible, but i dont know what amount of recoding this would take.

RoboDurden, 2022-07-23:
For a Linux believer a make file should be no issue at all. But I am rather a Linux hater.
Yet if you tell me how to install PlatformIO on my Ubuntu vserer and which exact command line will successfully compile the firmware, I would add platform Io support to my online compiler :-)

BastelPichi, 2022-07-24:
Well to install PIO, you get quite good docs here: [https://docs.platformio.org/en/latest/core/installation/methods/index.html](https://docs.platformio.org/en/latest/core/installation/methods/index.html)

To compile the project, all you need to do is run `pio run` in the `HoverBoardGigaDevice`directory.
The build is saved at `HoverBoardGigaDevice/.pio/build/GD32F130C8T6/` theres a firmware.bin and a firmware.elf, I assume they are both the same, but I would try to firmware.elf first.

The first build takes quite some time as it downloads alot, however then builds run within a few seconds.

PS: Lol ich könnte auch einfach Deutsch schreiben
[image]

BastelPichi, 2022-09-03:
Any update on this?

RoboDurden, 2022-09-03:
no time (time = happiness) to add the PIO support. sorry.

BastelPichi, 2022-09-04:
Ok, I broke my STLink anyways for now xD

weiminshen99, 2023-04-24:
Here is a repository for Makefile, if you are still interested:
https://github.com/weiminshen99/GD32-hover-master-slave

RoboDurden, 2023-04-24:
Yes great,  i still would like to add a gen2 firmware to my online compiler.
Your folder structure

https://github.com/weiminshen99/GD32-hover-master-slave/tree/main/Drivers/GD32F1x0_standard_peripheral/Source

 is a bit different then my gen 2.x firmware

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/HoverBoardGigaDevice/RTE/Device/GD32F130C6

But I guess I can port your makefile.

If your makefile generates a working gen2.0 binary i would add your report to my online compiler. It might take a year before I proceed with the makefile for my repo...

RoboDurden, 2023-04-24:
no but i have made quite a lot of reverse engeniering: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/Schematics_2.0

RoboDurden, 2023-04-26:
So i have added your repo to my online compiler. But i forked it because i needed to switch to a gcc inline compiler directive. Still get a warning 

```
Src/bldc.c:90:38: warning: always_inline function might not be inlinable [-Wattributes]
   90 | __attribute__((always_inline))  void blockPWM(int pwm, int pwmPos, int *y, int *b, int *g) //org __INLINE not known in gcc

```
but the code compiles:  https://pionierland.de/hoverhack/?p=gen2
Maybe someon can test it.

I also needed to increase a char buffer from 5 to 7 in comsBluetooth.c to remove sprintf warnings:

```
void SendBluetoothDevice(uint8_t identifier, int16_t value)
{
	int index = 0;
	char charVal[7];

```

This should also apply to your compiler ?

Thanks again.

weiminshen99, 2023-04-26:
Thanks for adding my repo!

I have made the two changes you suggested and compiled here successfully and pushed the changes into my repo.

If you like, you could try to compile again to see if you can get rid of the warming (replace “_INLINE” by “inline”).

Have a good day!

RoboDurden, 2023-04-27:
Yes i would like to add your original and not my fork so you might be motivated to get involved.
But I had already tried the official c inline and it does.not compile:

/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: build/bldc.o: in function `CalculateBLDC':
/var/www/html/pionierland/public/hoverhack/source/robodurden_gd32-hover-master-slave/Src/bldc.c:214: undefined reference to `blockPWM'
collect2: error: ld returned 1 exit status
make: *** [Makefile:185: build/firmware.elf] Error 1

I can remove the inline altogether and i compiles without warning. Is that function really that time critical that there is no time to make a function call ?

RoboDurden, 2023-04-27:
I See that the blockPWM function is used only once anyway : https://github.com/weiminshen99/GD32-hover-master-slave/blob/main/Src/bldc.c#L214

So simply copy the code there ?

I think it should be possible to make the function code a #define , that might be another was to get rid of that inline word ?

weiminshen99, 2023-04-27:
I have taken out the “__INLINE”, and please compile again. Thanks!

RoboDurden, 2023-04-27:
Fine, i switched the repo back to yours :-)

https://pionierland.de/hoverhack/?p=gen2

Would be nice if you could add a simple arduino example that spins the two motors a bit forward and backwards.
This seems to be too much: https://github.com/gaucho1978/CHEAP-LAWNMOWER-ROBOT-FROM-HOVERBOARD/tree/master/ArduinoFirmware

For my Gen 2.x firmware i rewrote all the serial communication, so people can not rely on my arduino example when using your firmware.
The original https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2/tree/master/PPMDeviceArduino_V1.0/PPMDevice is also overloaded with PPM rc-control :-/

Ideas welcome :-)

weiminshen99, 2023-04-28:
Please open an issue at the repos: https://github.com/weiminshen99/GD32-hover-master-slave
and I will add something as you requested there. Thanks!