flo199213/Hoverboard-Firmware-Hack-Gen2 #12: Error flashing with ST-LINK utility

JMarceloCasanova, 2020-04-22:
Hello! I'm having an error flashing the firmware with ST-LINK.

So what I have done is: compiled the firmware with Keil to have the binaries, connected the st-link v2 with the board and unlocked the flash by resetting the option bytes, cleared the memory and then I get this error:
: Unexpected error
: Error occured during program operation!
: Programming error @ 0x08000000!
I'v tried pressing the button and changing some speeds but with no success.

My board's model is TT-SD2.2 with a gd32f130c8t6

andresiko, 2020-07-22:
Any progresa @JMarceloCasanova ?

drphil3d, 2022-02-02:
having the same issue

rokars, 2022-06-06:
i used the keil flash -> download to program directly from keil, works

AlfreddGco, 2023-12-01:
Try with `st-flash --reset write ./firmware.bin 0x08000000`