flo199213/Hoverboard-Firmware-Hack-Gen2 #14: ADC control with two potentiometers

vischio96, 2020-05-02:
Hi all,

these two identical mainboards are pretty common in hoverboards, and i have a lot of them at home. Does anyone know if there's a way to control the two boards with 2 differents ADC inputs (ex two potentiometers) as the mainboard that NiklasFauth uses? 

From the reverse engineered schematics i see that the two lightbarriers sensors uses two ADC pins (PA7 and PA11) and they seems unused in the code, but i'm not that good at programming. 
The only way i know to control these mainboards is using PPM with an Arduino board.

Could anyone help me? Thank you a lot.

andresiko, 2020-07-22:
Have you succeeded flashing these boards? I have an tt-sd 2.2 with gd32f130c8t6, and I don't know if this firmware will work on it.

zulkep, 2021-06-16:
have you got to connect the potentiometers directly to the board?

Jebumon, 2022-04-25:
Hi all, 

I have hoverboard control board, using GD32F130C8T6. Anyone know how to re-program his MCU ? 
could anyone help me ?

alexei0kazachina, 2023-06-02:
Да пишите мне, я смогу помоч.