flo199213/Hoverboard-Firmware-Hack-Gen2 #15: No compiling , error in adc

centiane, 2020-08-31:
No compiling ok, this error show, help me, please.

*** Using Compiler 'V5.06 update 6 (build 750)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Rebuild target 'Target 1'
compiling setup.c...
Src\setup.c(343): error:  #167: argument of type "dma_parameter_struct" is incompatible with parameter of type "dma_parameter_struct *"
  	dma_init(DMA_CH0, dma_init_struct_adc);
Src\setup.c(365): error:  #20: identifier "ADC_EXTTRIG_REGULAR_SWRCST" is undefined
  	adc_external_trigger_source_config(ADC_REGULAR_CHANNEL, ADC_EXTTRIG_REGULAR_SWRCST);
Src\setup.c(424): error:  #167: argument of type "dma_parameter_struct" is incompatible with parameter of type "dma_parameter_struct *"
  	dma_init(DMA_CH4, dma_init_struct_usart);
Src\setup.c(480): error:  #167: argument of type "dma_parameter_struct" is incompatible with parameter of type "dma_parameter_struct *"
  	dma_init(DMA_CH2, dma_init_struct_usart);
Src\setup.c: 0 warnings, 4 errors

cfox570, 2020-10-18:
It looks like the uVision IDE has new libraries that are required.  I am rebuilding the project with uVision and will add the source to it.  I will update my comment when I have additional results.

mmaslyukov, 2021-01-25:
> No compiling ok, this error show, help me, please.
>
> *** Using Compiler 'V5.06 update 6 (build 750)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
> Rebuild target 'Target 1'
> compiling setup.c... [...]

Recently tried to compile it, but got the same errors. However after downgrading GigaDevice pack to 3.1.0 and going through steps mentioned in other threads I was able to successfully build it. 
Here is a brief list of neccessery steps to do 

1. Use  GigaDevice.GD32F1x0_DFP.3.1.0.pack, version 3.2.0 will not do 
2. In uVision project settings (C/C++ tab) add "C:\Users\<your_user_name>\AppData\Local\Arm\Packs\ARM\CMSIS\5.7.0\CMSIS\Include" to include path
3. In uVision project settings (Linker tab)  set checkbox: "Use Memory Layout form Target Dialog". It is necessary for generation of  Hoverboard.sct file 
4.  Compile

Dennis-Gee, 2021-06-08:
Hello, the GigaDevice.GD32F1x0_DFP.3.1.0.pack is no longer available. could someone compile the two firmwares and upload ist somewhere so the project isnt dying?

cfox570, 2021-06-08:
You need to get GD32F1x0_Addon_V3.1.0.rar from  http://www.gd32mcu.com/en/download/7/p/2   Use the pack installer to downgrade to 3.1.

rokars, 2022-06-06:
3.1.0 pack available here https://github.com/gearsincorg/Hoverboard_Utility_Gateway_System

alexismailov2, 2024-08-11:
using 3.1.0 fixed previous problems but. Apeared additional problem 
.\Objects\Hoverboard.axf: error: L6031U: Could not open scatter description file .\Objects\Hoverboard.sct: No such file or directory
Could someone help?

IrokezPetrovic, 2025-06-02:
In SPL 3.10:
`#define ADC_EXTTRIG_REGULAR_SWRCST      CTL1_ETSRC(7)                       /*!< software trigger select for regular channel */`
In newer versions of SPL:
`#define ADC_EXTTRIG_REGULAR_NONE        CTL1_ETSRC(7)                       /*!< software trigger */`

To use SPL newer than 3.1.0 - replace `ADC_EXTTRIG_REGULAR_SWRCST` to `ADC_EXTTRIG_REGULAR_NONE`