flo199213/Hoverboard-Firmware-Hack-Gen2 #17: How to flash?

casperround, 2021-03-23:
When flashing do you have to flash both boards? I've flashed the main master successful, powered on and the slave board operates like stock. So wondering if I need to flash that side too??

FGay, 2021-05-01:
you need to flash both sides but you have to rebuild the code with a change in the config.h file
for the master board you can leave everything as is but for the slave board you need to change "#define MASTER" to "//#define MASTER" and also you have to uncomment the "#define SLAVE"

szepnorbee, 2021-05-08:
> you need to flash both sides but you have to rebuild the code with a change in the config.h file
> for the master board you can leave everything as is but for the slave board you need to change "#define MASTER" to "//#define MASTER" and also you have to uncomment the "#define SLAVE"

Hi FGay! Have you tried this firmware? I tried Pro Mini and ESP8266, but no response. I try to access it on MASTER using the example code.(PPMDevice) The controller sends the data, the arduino sends the response, but nothing. The board is good because I can start the motors at a fixed speed by modifying the code (board firmware). Can you help me?  Thank you.

FGay, 2021-05-09:
@szepnorbee 
i also have issues.
The board sends the data properly. My arduino nano/uno is not capable of listening to the board on its RX pin. (dont know why though)
I tried using software serial and now the communication between the arduino and the hoverboard works.
But the motors are not moving. I also tried to spin the motors at a constant speed by modifying the board firmware but i was not able to get it to work. (the hoverboard was perfectly fine before the flash so i dont think that its an hardware issue)
I also used the example code but modified it slightly so i can adjust the speedvalue with a potentiometer.
I think we might have a similar problem here.
How does your modified board firmware look like?

szepnorbee, 2021-05-09:
> @szepnorbee
> i also have issues.
> The board sends the data properly. My arduino nano/uno is not capable of listening to the board on its RX pin. (dont know why though)
> I tried using software serial [...]

Yes, I have solved the problem. There were two things. The first was that my board had a bluetooth interface which is not shown in the photo. I soldered it down because it was occupying the USART ports. The second problem is that the arduino code called PPMDevice is broken. It originally sends 9 bytes instead of 8. The one line in SteeringSerial.cpp needs to be commented out. This: // buffer[index++] = sendByte;

After doing this, everything works immediately. 

I use this fork now because it has PID and you can control the two motors separately: https://github.com/pieterjanbuntinx/Hoverboard-Firmware-Hack-Gen2

Good luck, I hope I helped.

Translated with www.DeepL.com/Translator (free version)

FGay, 2021-05-10:
I am so happy. IT WORKS NOW!! 
Thank you very much!

rohitsahu7, 2021-07-31:
Hello Great to hear ..congrats for your project . I need help regarding the project compiling in Keil , I am getting errors, i have  not modified anything still getting errors basically in timers,
Please help , i have attached picture
thank you 

[image]