flo199213/Hoverboard-Firmware-Hack-Gen2 #20: What accelerometer/gyroscope chip does the board actually have?

qwandor, 2021-10-02:
The schematic here says that there's an MPU6050 connected to I2C2 on PB8/PB9, but this seems not to be accurate. For a start PB8/PB9 is I2C0 not I2C2, but more importantly there doesn't seem to be any MPU6050 on the board.

There are however two chips that I haven't been able to identify: U2 and U5. I'm struggling to read the markings on them, but U2 is marked something like "ATG BS." while U5 is marked "TP09. 1828 H1" or something similar. Has anyone been able to figure out what these chips are, and whether one of them is the sensors?

qwandor, 2021-10-02:
To confuse matters, looking at another board, the chips are different: on this one U2 is "BOP BS." and U5 is "TP09. 1822 H1".

Candas1, 2021-11-05:
Hi,

It could be there are different versions of the board with different IMUs also.