flo199213/Hoverboard-Firmware-Hack-Gen2 #21: ARM MM32SPIN06 YS-SXT-4.2 - HOVER-1 Board

landgrant, 2022-01-21:
I recently purchased a HOVER-1 CHROME hoverboard. Inside it had the split boards similar to the main post but has a different processor I have not seen. I cannot get the board to connect on ST-LINK, and not sure if the pinout is the same as others have seen with their split boards. The main board is labeled YS-SXT-4.2 A, the slave is YS-SXT-4.2 B. I installed the MDK ARM tool from KEIL, but STM32Cube and ST-LINK still have no luck connecting.

https://imgur.com/a/8aWoPEG

I haven't seen anything online with info, but wanted to see if anyone had any tips.

robinsingh1, 2022-04-25:
Were you able to find an answer here ?

landgrant, 2022-04-25:
Unfortunately not. I ended up buying single boards off of eBay for one
project, and separate brushless controllers for another

Kambodia91, 2023-04-11:
look: https://github.com/Kambodia91/Hoverboard-firmware-hack-FOC-MASTER-SLAVE

Alshawa994, 2024-09-05:
I'm working on the same Motherboards, and this can help you. It is also helpful to know which Processor you have on it. mine was MM32SPIN05PF on a YS-SXT-V4.2 A & B.

here is the link 
https://github.com/AILIFE4798/Hoverboard-Firmware-Hack-Gen2.x-MM32/blob/main/README.md

omarmahamed346-art, 2025-11-15:
I have the same board and I am stuck trying to get the pins finder working. HELP PLS

USERDK0, 2026-10-21:
YS-SXT-4.2 B Board:

[image]
[image]

Kambodia91, 2026-10-25:
YS-SXT-4.2 A Board:
[image]
[image]
[image]
[image]