flo199213/Hoverboard-Firmware-Hack-Gen2 #26: RC Receiver

WolfgangDerVollstrecker, 2023-01-23:
any way to control the split board with PWM from an RC Receiver?

RoboDurden, 2023-01-30:
I guess we neeed to help ourselves..
this firmware seems to only support uart control. So better add a $3 ESP32 S2 Mini to your project which will add bluetooth and wifi, etc.

Of course you could add code to read a pwm signal from the RX pin.
Speed and steer are set in https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2/blob/e138b221aed944478d12e15d81fb858f44f4a289/HoverBoardGigaDevice/Src/commsSteering.c#L128

And here you can find the code to implement pwm: https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/efb438bf514c9f8c80471f79ab87d2f7099746ea/Src/control.c#L173

But i strongly suggest you add the esp32..

ADC with two potentiometers for speed and steer would also be much simpler. The rx and tx pins (PB6 and PB7)  however offer no adc (https://raw.githubusercontent.com/flo199213/Hoverboard-Firmware-Hack-Gen2/master/GD32F130C8%20Dokumentation/GD32F1x0_User_Manual.pdf page 16.)

But some led outputs could be configured as analog input (PA0 and PA1 for instance). You can find the accessible pins in inc/defines.h.  

Have you succeeded to compile this firmware Gen2.0, flash it and the motors can be controlled via uart ?
If so, please send me detailed photos of your two control boards. I try to build a Gen2.x firmware and my pin assignemnts are totally different from what this firmware, yet the photo of the two circuit boards is identital to my board: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x

RoboDurden, 2023-02-01:
You can now download compiled binaries for the V2.0 and V2.1 boards as well as tested Arduino examples on my fork: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x

I can only test the PPM rc-receiver Arduino example with OVERWRITE_PPM as i do not have an rc-controller. Please share feedback.

Vorlovskis, 2023-06-04:
> You can now download compiled binaries for the V2.0 and V2.1 boards as well as tested Arduino examples on my fork: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x
>
> I can only test the [...]

Hi. Can you share your experience with versions 2.0 dual mainboards?   stuck in visual studio error in LN34 no fail sourse found. Please help

RoboDurden, 2023-06-04:
i use the Keil ide for my gen2.x repo. PlatformIO uses a different tool chain so compiler errors might occour.
Right now we are trying to build a PlatformIO C++ SimpleFOC firmware for the GD32F130 Gen2 boards here: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/11
As the GD32-Arduino-Core is still very buggy it is yet to be seen how easy that will be.
But if you have a Gen2.0 board or Gen1 Sideboard with GD32F130 you are very welcome to join the party.

Roland, from www.RoboDurden.de

Vorlovskis, 2023-06-04:
Thanks for reply. I am basicaly 0 in coding with basic in Arduino. If i can help with smth, just tell. My dual motherboards are GD32F130C8. One set 2.0, second is bit different after chip pinout to mosfets. But pins still same. Looking to use PWM Receiver.

RoboDurden, 2023-06-04:
for the 2.0 you can user my precompiled binaries. You will need an esp32 (hardware serial) or esp8266/arduino (software serial) anyway.